ImpulseAdventure/Waveshare_ILI9486 #1: Text size smaller than expected

deepu2, 2019-10-28:
**Describe the bug**
I'm using the Waveshare_ILI9486 4" display along with your library. Installed libraries are:

1. Adafruit_GFX_Library
2. Adafruit_ILI9341
3. GUIslice
4. SD
5. Waveshare_ILI9486
6. XPT2046_Touchscreen

GUIslice Builder ver 0.13.0.3
In options, TFT WxH is set as 480x320

Now whenever I use texts on my display regarding of the setting it is displayed too small.

**Expected behavior**
The size of the text should have been bigger.

**Crash log / Project files**
[ds.zip](https://github.com/ImpulseAdventure/GUIslice-Builder/files/3776040/ds.zip)

**Screenshots**
[This is how it looks in the builder](https://imgur.com/I362vxu)
[This is how it looks in the display](https://imgur.com/suEe7OH)

**Version**
- Builder version: 0.13.0.3
- OS: Windows 10

deepu2, 2019-10-28:
Hello, thank you for the response. I can help test it :)

ImpulseAdventure, 2019-10-31:
@deepu2 -- I have now updated GUIslice to support built-in Waveshare font selection via setFont(). This means that you can change the font sizing from within the GUIslice Builder and see the new fonts on the display. To use these waveshare fonts in the Builder, you will need to update the "arduinofonts.csv" file within your Builder install directory.

Step 1:
- Update GUIslice to the latest version (0.13.0.8) in the GitHub repository

Step 2:
- Locate the `arduinofonts.csv` file in your Builder install directory. eg. `C:\Program Files (x86)\GUIsliceBuilder\templates\arduinofonts.csv`.
- Insert the following lines (eg. after the `BuiltIn(5x)` line):
```
Waveshare5x8,Waveshare_ILI9486_GFX.h,NULL,E_FONT_WS8,GSLC_FONTREF_PTR,&Font8,1,Monospaced,5,PLAIN
Waveshare7x12,Waveshare_ILI9486_GFX.h,NULL,E_FONT_WS12,GSLC_FONTREF_PTR,&Font12,1,Monospaced,7,PLAIN
Waveshare11x16,Waveshare_ILI9486_GFX.h,NULL,E_FONT_WS16,GSLC_FONTREF_PTR,&Font16,1,Monospaced,11,PLAIN
Waveshare14x20,Waveshare_ILI9486_GFX.h,NULL,E_FONT_WS20,GSLC_FONTREF_PTR,&Font20,1,Monospaced,14,PLAIN
Waveshare17x24,Waveshare_ILI9486_GFX.h,NULL,E_FONT_WS24,GSLC_FONTREF_PTR,&Font24,1,Monospaced,17,PLAIN
```

Step 3:
- Reopen the Builder and select the new "Waveshare" fonts
- Regenerate code and upload

Let me know if that works better.

deepu2, 2019-10-31:
The font looks great!! [image]
But the touch has been ruined(?) it has rotated 180 degrees

deepu2, 2019-10-31:
The touch must've been flipped I think, touch is working as portrait mode(Sorry if this doesn't help).

ImpulseAdventure, 2019-10-31:
Can you run the touch calibration utility (diag-ard-touch-calib) and report the output for me? This utility should indicate the changes needed for your config to match your particular display / touch setup. Thanks!

deepu2, 2019-11-01:
```
GUIslice version [0.13.0.8]:
- Init display handler [WAVESHARE_ILI9486] OK
- Init touch handler [XPT2046_PS(SPI-HW)] OK

=== Touch Calibration ===

CALIB: Config defaults: XMin=234 XMax=3968 YMin=171 YMax=3823 RemapYX=1

CALIB: Averaging mode: BASIC

Capture TL: X=311 Y=356 Cnt=40, Z=(1881..2439)
Capture TR: X=379 Y=3855 Cnt=35, Z=(1123..1417)
Capture BL: X=3726 Y=348 Cnt=51, Z=(2165..2241)
Capture BR: X=3782 Y=3862 Cnt=57, Z=(400..435)
SegDetect: RemapYX display
CALIB: Result(adj) XMin=235 XMax=3974 YMin=270 YMax=3828

CALIB: Rotate
Capture TL (Rotated): X=345 Y=294 Cnt=41, Z=(2007..2537)

Recommended calibration settings are shown below.
- Copy and paste these lines into your GUIslice config file
  over top of the existing ADATOUCH_X/Y_MIN/MAX settings.

---
  // DRV_TOUCH_XPT2046_PS [320x480]:
  #define ADATOUCH_X_MIN    235
  #define ADATOUCH_X_MAX    3974
  #define ADATOUCH_Y_MIN    270
  #define ADATOUCH_Y_MAX    3828
  #define ADATOUCH_REMAP_YX 1
  #define GSLC_ROTATE_SPECIAL 3 // No Rotation supported
---

CALIB: Apply calibration

```

This is from the latest GUIslice from github. 
In the confirm red dot match touches , the dots aren't matching.

deepu2, 2019-11-01:
```
GUIslice version [0.13.0]:
- Init display handler [WAVESHARE_ILI9486] OK
- Init touch handler [XPT2046_PS(SPI-HW)] OK

=== Touch Calibration ===

CALIB: Config defaults: XMin=234 XMax=3968 YMin=171 YMax=3823 RemapYX=1

CALIB: Averaging mode: BASIC

Capture TL: X=273 Y=340 Cnt=55, Z=(2001..2657)
Capture TR: X=280 Y=3877 Cnt=37, Z=(1321..1501)
Capture BL: X=3755 Y=339 Cnt=50, Z=(2207..2367)
Capture BR: X=3770 Y=3856 Cnt=78, Z=(407..546)
SegDetect: RemapYX display
CALIB: Result(adj) XMin=262 XMax=3942 YMin=159 YMax=3878

CALIB: Rotate
Capture TL (Rotated): X=337 Y=308 Cnt=65, Z=(2025..2577)

Recommended calibration settings are shown below.
- Copy and paste these lines into your GUIslice config file
  over top of the existing ADATOUCH_X/Y_MIN/MAX settings.

---
  // DRV_TOUCH_XPT2046_PS [480x320]:
  #define ADATOUCH_X_MIN    262
  #define ADATOUCH_X_MAX    3942
  #define ADATOUCH_Y_MIN    159
  #define ADATOUCH_Y_MAX    3878
  #define ADATOUCH_REMAP_YX 1
  #define GSLC_ROTATE_SPECIAL 3 // No Rotation supported
---

CALIB: Apply calibration
```
This was the older one v0.13.0

ImpulseAdventure, 2019-11-01:
Thanks for providing the details.

Q1) The utility captures the 4 corners (TL, TR, BL, BR), attempts a rotation and then captures the TL corner again (rotated). Do you see the screen rotate after the BR corner?

Q2) If you continue to run the test mode (after calibration where it shows red dots), it is supposed to rotate the screen every time you stop pressing on the screen. Do you see the screen rotate?

From your log file it seems like your 4" Waveshare display might not be rotating properly. I don't recall seeing this happen before.

Q3) You could try forcing a manual calibration to see if we can match the orientation of your display. For example, if you were seeing a 180 degree mismatch, then we could try swapping the coordinates in your config:
```c++
  #define ADATOUCH_X_MIN    3974 //235
  #define ADATOUCH_X_MAX    235 //3974
  #define ADATOUCH_Y_MIN    3828 //270
  #define ADATOUCH_Y_MAX    270 //3828
  #define ADATOUCH_REMAP_YX 1
  //#define GSLC_ROTATE_SPECIAL 3 // No Rotation supported
```
After making the above changes to your config, you can run `diag-ard-touch-test` to see if the red dots line up. It may take some experimentation to determine which calibration values (X_MIN/MAX, Y_MIN/MAX) might need swapping. Does this change the mapping to match the red dots?

deepu2, 2019-11-01:
Q1) I didn't notice the screen rotating, the text was still in the same layout.
Q2) the screen didn't rotate here too. It didnt rotate when i ran it with the GUIslice version in arduino.
I'll try Q3 and update.

deepu2, 2019-11-01:
I was about to give up but tried one last time with the latest ILI9486 library and ran the caliberation. 
It was successful, screen was rotating.

ImpulseAdventure, 2019-11-01:
Great! Yes, I had forgot to mention that the Waveshare dynamic rotation feature was only added recently to the repository, so that would explain why the rotation wasn't working :)

I think with the latest Waveshare_ILI9486 and GUIslice libraries, we should now have the ability to use alternate sized fonts in the GUIslice builder as described above. I will close this now, but if there are any additional issues, please feel free to open a new ticket.

As for Adafruit-GFX fonts, I am going to defer that for the moment as Waveshare provided alternate-sized fonts (as mentioned in this issue). Providing full access to GFX-style custom fonts is a much larger enhancement that I could consider again later.

thanks for the feedback and testing!

closed 2019-11-01 by ImpulseAdventure

pitrz, 2019-11-18:
Hi,

i've been playing around with the new version and have noticed a strange behavior with text position / alignment. Basically what is happening if i use the waveshare font as per instructions is that alignment works as it supposed to when the element is stored in RAM. If use the PROGMEM / flash api / _P versions of the element (Button) labels are not aligned correctly. Can you please give me a hint what could be off? 

Also - the code is generated using the builder.

Thanks,

Petr

ImpulseAdventure, 2019-11-19:
> Hi,
>
> i've been playing around with the new version and have noticed a strange behavior with text position / alignment. Basically what is happening if i use the waveshare font as per instructions is [...]

Hi @pitrz -- thank you for reporting this. A change needs to be made in GUIslice to address this issue. I have moved your issue to the GUIslice repository here: https://github.com/ImpulseAdventure/GUIslice/issues/173