ImpulseAdventure/Waveshare_ILI9486 #3: Is ESP32 planned here?

szymoni, 2019-10-28:
Hello,

as I can see, the ESP32 is not taken here into consideration, or?
- .\Waveshare ILI9486_ID6670\src\Fonts\font16.c:40:26: fatal error: avr/pgmspace.h: No such file or directory
- .\Waveshare ILI9486_ID6670\src\DEV_Config.cpp:43:28: error: 'analogWrite' was not declared in this scope
   analogWrite(LCD_BL, value);

I would really like to see this TFT working on my ESP32. In PlatformIO's library description is mentioned as Espressif32 compatible...

Thanks,
SID

szymoni, 2019-10-28:
>>> The problem with FONTs is solved ->  instead of #include <avr/pgmspace.h>, used #include <pgmspace.h> 
>>> The lacking definition of  analogWrite() for ESP32 in Arduino Lib, could be substituted with external LIB - [](https://github.com/ERROPiX/ESP32_AnalogWrite.git)

... but there is more problems on compile now.

ImpulseAdventure, 2019-10-30:
Hi szymoni --

As you have discovered, the original Waveshare library was intended for the Arduino shield and isn't likely to compile without a number of modifications for the ESP32.

I strongly suggest you give Bodmer's excellent [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) library a try as I understand it should be able to drive the Waveshare display on the ESP32 platform. His library is very well optimized for Espressif devices and should deliver much better performance. Note that I haven't tried running my Waveshare TFTs with TFT_eSPI yet.

If you do use TFT_eSPI, I believe you might want to try uncommenting the following line in your User_Setup_Select file: `#include <User_Setups/Setup11_RPi_touch_ILI9486.h>`

Hope that helps!

ImpulseAdventure, 2020-01-08:
`library.properties` has been updated to limit the scope to AVR architectures.

closed 2020-01-08 by ImpulseAdventure