ImpulseAdventure/Waveshare_ILI9486 #5: Missing function GET_TP_IRQ?

DavidVanHorn, 2020-01-01:
I may be doing something wrong here, but in compiling your library into my project, I've edited the DEV_Config.h to include the pins for touchpanel and SD card and added the appropriate macros for taking them high and low.

I now have one error remaining,  I don't seem to have the function GET_TP_IRQ anywhere. 
It's used in TP_Scan as below.

Can you tell me what I'm doing wrong?

`
/*******************************************************************************
  function:
        Calculation
  parameter:
        chCoordType:
                    1 : calibration
                    0 : relative position
*******************************************************************************/
static unsigned char TP_Scan(unsigned char chCoordType)
{
  //In X, Y coordinate measurement, IRQ is disabled and output is low
  if (!GET_TP_IRQ) {//Press the button to press
    //Read the physical coordinates`

Arduino: 1.8.10 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Dave\Documents\Arduino\libraries\4inch_TFT_Touch_Shield_Code\LCD_Touch.cpp: In function 'unsigned char TP_Scan(unsigned char)':

C:\Users\Dave\Documents\Arduino\libraries\4inch_TFT_Touch_Shield_Code\LCD_Touch.cpp:152:8: error: 'GET_TP_IRQ' was not declared in this scope

   if (!GET_TP_IRQ) {//Press the button to press

        ^~~~~~~~~~

C:\Users\Dave\Documents\Arduino\libraries\4inch_TFT_Touch_Shield_Code\LCD_Touch.cpp:152:8: note: suggested alternative: 'TP_IRQ'

   if (!GET_TP_IRQ) {//Press the button to press

        ^~~~~~~~~~

        TP_IRQ

Multiple libraries were found for "DEV_Config.h"
 Used: C:\Users\Dave\Documents\Arduino\libraries\4inch_TFT_Touch_Shield_Code
Multiple libraries were found for "SPI.h"
 Used: C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\SPI
Multiple libraries were found for "Wire.h"
 Used: C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\Wire
Multiple libraries were found for "EEPROM.h"
 Used: C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\EEPROM
Multiple libraries were found for "SD.h"
 Used: C:\Users\Dave\Documents\Arduino\libraries\SD
 Not used: C:\Program
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

ImpulseAdventure, 2020-01-07:
Hi David -- sorry for the delay & thanks for the report.

The `GET_TP_IRQ` function is actually a macro defined in [DEV_Config.h](https://github.com/ImpulseAdventure/Waveshare_ILI9486/blob/8d44112520bf5d88253547eb5af40765ddbce2d2/src/DEV_Config.h#L40)
[image]

Is it possible that you might have inadvertently removed this line when editing your DEV_Config?

DavidVanHorn, 2020-01-07:
Looks like I did.  Thanks!  User headspace errror strikes again :)

ImpulseAdventure, 2020-01-08:
Glad to hear that was it! Thanks for the followup

closed 2020-01-08 by ImpulseAdventure