ImpulseAdventure/Waveshare_ILI9486 #7: Current version not compatible with GUISlicer

sjbaird, 2020-06-16:
It seems that this version is missing the GFX files

ImpulseAdventure, 2020-06-17:
Hi! You are correct that the updated version has changed set of include files.

A new version of GUIslice has been released that uses the updated Waveshare_ILI9486 files.
This version is not yet released via the Arduino IDE library manager, so for now you will need to [manually update the library from GitHub](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino#installing-from-github-source).

Let me know if you have any difficulties.
Thanks!

ImpulseAdventure, 2020-08-24:
Closing as GUIslice 0.15 has been released to the Arduino IDE library manager

closed 2020-08-24 by ImpulseAdventure