ImpulseAdventure/Waveshare_ILI9486 #9: ESP32 or Wemos D1 R32 - ILI9486 4"

pawelmielnik, 2021-01-11:
Hi  Calvin Hass and M Hotchin, 
I used your src to run 9486 on ArduinoMega and ESP32 and Wemos. On Mega everything is ok.
On ESP32 I have "blank screen" serial port shows communication. 
On Wemos screen is also blank, but on serial port showes rubbish  In cpp i changed the analogWrite LCD_BL to digitalWrite, it doesn't help at all. 
Have you ever met such issues.
Hardware is ok, runs on other applications.
Pawel,

StefanKellerAC, 2022-01-21:
I had a white screen with a too high SPI_FREQUENCY in Setup11_RPi_touch_ILI9486
with an ESP32, an 3.5 inch Raspi Display (from ELEGOO) and ILI9486_DRIVER
with
#define SPI_FREQUENCY  15000000
it works

Mega-mania, 2022-01-21:
Great,

Thanks for your response.  I was in a tough spot, but eventually decided to
code my own very low ram fast gui which is working with 356 bytes of ram
with menus, responding buttons, layering etc. Refreshing is by group, which
is not as neat and elegant as others, but good enough for my use.

Kind regards,

Teo

> I had a white screen with a too high SPI_FREQUENCY in
Setup11_RPi_touch_ILI9486
> with an ESP32, an 3.5 inch Raspi Display (from ELEGOO) and ILI9486_DRIVER
> with
> #define SPI_FREQUENCY 15000000
> it works
https://ci4.googleusercontent.com/proxy/OfrP5Et9BUCKYYbjeOelV8MwM8WkEUYbTh2PiEGyCfwkNvKqNwUSlRWol2eRFo1ktfbvwrEnN7p9E86nmkO8wmvCesiEh6fP7uvX2Mk7shiF_Jl8X4yKGtzUem3I5auvbM1yHuha71wU6ieg2d6kjQ5e8vxgWfhnAQsOlDYAMtb6ChkXTkmZwNHE-rYnLICYFEe6TMa7jX7Jd5QAvvvwVGSqUsVhYlYUGRXAgpj86w=s0-d-e1-ft#https://github.com/notifications/beacon/AVAENJXNX5JBTMPTMCUHUGLUXFK4RA5CNFSM4V5ICJA2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOHS2OKMA.gif>Message
ID: ***@***.***>