ImpulseAdventure/Waveshare_ILI9486 #11: Arduino Due

Ricky1966, 2021-09-01:
Hi, I'm sorry for posting my question as an issue, but I ain't know other way, I want to know if this library work with Arduino Due and if (and how) is possible to change pins for using for example pin 30 and more.
Tks

Reisen-Wandern-Tauchen, 2022-04-08:
Did you find any solution on changing the pins so far?
I'd like to use two displays in the same project so would need to redefine the CS-pin for one of them...

Mega-mania, 2022-04-08:
Hi,

No, I eventually decided to code my own low resource GUI.

Regards,

Teo

> Did you find any solution on changing the pins so far?
> I'd like to use two displays in the same project so would need to
redefine the CS-pin for one of them...
https://ci3.googleusercontent.com/proxy/nKqFktB4sdzaQKQbfHbl4lX29l_8CNU_HUiXbDPAsJKsVoxAhe5GRZllIgvK65ioHt2T4Wyj4bRirQtc5erkVu3WnVU6_aepgNlw1M1okoiG9fpWc7gatOkSn-gm5NYNlMGJncbQhRDWexdwZkXRnztFPu4m1KhK8LY7ILCzTfaGjwpsCJ8Oz6mJ4NdXkHvkI7v-lcJqdrpk1PcoksT7NWjgZnxkxJ6ygdsr4OkZDQ=s0-d-e1-ft#https://github.com/notifications/beacon/AVAENJWNSHUJSEUUJI3D3IDVD7IB3A5CNFSM5DF4WQU2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOIEPCJXI.gif>Message
ID: ***@***.***>